3dgeo-heidelberg/helios #373: Multiple values in scannersettings in survey.xml

Hannes9, 2023-09-18:
Is it possible to set multiple values for the `horizontalResolution_deg`, `scanFreq_hz` , `headRotatePerSec_deg` in the scanner settings in the survey. xml file depending on the `scanFreq_hz `?

E.g. like that:

`horizontalResolution_deg="0.09, 0.18, 0.36" scanFreq_hz="5, 10, 20" headRotatePerSec_deg="1800, 3600, 7200" />`

Thanks a lot.
Hannes

lwiniwar, 2023-09-18:
Hi @Hannes9 ,

no, this syntax is not supported. What would be the expected behavior? Three separate simulations with the three sets of settings, respectively?

Cheers,
Lukas

Hannes9, 2023-09-18:
Hi,

Thanks for the info. I think I got the answer already by my own.
Probably I have to use the highest `scanFreq_hz` and its depending other parameters (mentioned before) for the getting the best results out of it.

Best,
Hannes

closed 2023-09-22 by lwiniwar