3dgeo-heidelberg/helios #374: Problems with initial scanFreq_Hz and headRotatePerSec_rad

Hannes9, 2023-09-19:
Hi,

I have a problem with keeping my initial ScanFreq_Hz and headRotatePerSec_rad. Which ratio should be set for the max effective scan angle and the max scan angle to be able to still use my initial settings ? As I got those initial values out of the manual of a LidDAR sensor, I am not sure how to set it up then correctly. My max effective scan angle and the max scan angle are set both to "360" in the scanner.xml.

I want to simulate the XT32M2X Hesai LiDAR which is a 360 degree mechanical lidar and the transmitters and receivers are rotating 360 degree.

Thanks in advance !

[image]

lwiniwar, 2023-09-20:
Hi @Hannes9,

please provide your `scanners.xml`, `platforms.xml`, and `survey.xml` so we can reproduce the behaviour on our side.

Cheers,
Lukas

Hannes9, 2023-09-21:
Hi @lwiniwar.

Sure. Here are the files that I use. For getting the simulation even executed I set the `scanFreq_hz` to 1200 Hz, but usually it should be 20 Hz.

Survey.xml:
```
<?xml version="1.0"?>
<document>
	<platformSettings id="platform1" movePerSec_m="20" onGround="true"/>
    <scannerSettings id="scanner1" active="true" verticalResolution_deg="1.3" horizontalResolution_deg="0.36" scanFreq_hz="20" headRotatePerSecMax_deg="7200" headRotatePerSec_deg="7200" headRotateStart_deg="0" scanAngle_deg="20.15" pulseFreq_hz="640000" trajectoryTimeInterval_s="0.01"/>
	<survey name="tum_data__-1_mls" platform="data/platforms.xml#navvis-vlx-3-on-person" scanner="data/scanners_als.xml#hesai_XT32M2X" scene="data/scenes/owndata/storeys/-1/storey_-1_scene.xml#tum_scene_-1">
	<!-- platform: person, deflector: rotating -->
		<leg>
			<platformSettings  template="platform1" x="2697493,772063" y="5349203,98595"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697486,589884" y="5349158,828849"/>
			<scannerSettings  template="scanner1"/>
		</leg>
	</survey>
</document>
```
Scanners.xml:

	<scanner 	id 							= "hesai_XT32M2X"
				accuracy_m					= "0.01"		
				beamDivergence_rad 			= "0.007224229598"
				name              			= "Hesai XT32M2X 32-Channel Medium-Range Mechanical Lidar"
				optics            			= "rotating"
				pulseFreqs_Hz				= "640000"
				pulseLength_ns     			= "30"
				rangeMin_m					= "0.5"
				rangeMax_m					= "300"
				scanAngleMax_deg   			= "360"
				scanAngleEffectiveMax_deg	= "360"
				scanFreqMin_Hz				= "5"
				scanFreqMax_Hz				= "20"
				wavelength_nm				= "905"> 				
				
		<FWFSettings beamSampleQuality="1"/>	
		<channels>
		
			<channel id="0">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="19.5" />
				</beamOrigin>
			</channel>
            <channel id="1">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="18.2" />
				</beamOrigin>
			</channel>
            <channel id="2">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="16.9" />
				</beamOrigin>
			</channel>
            <channel id="3">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="15.6" />
				</beamOrigin>
			</channel>
            <channel id="4">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="14.3" />
				</beamOrigin>
[... 349 lines cut ...]
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="19.5" />
				</beamOrigin>
			</channel>
            <channel id="63">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="20.8" />
				</beamOrigin>
			</channel>
			
        </channels>
	</scanner>
Platforms.xml:
`	<platform id="navvis-vlx-3-on-person" name="NavVis VLX 3 on person" type="linearpath">
		<scannerMount x="0" y="0" z="0">
			<rot axis="x" angle_deg="0" />	
		</scannerMount>
		
	</platform>`

Thanks a lot.

lwiniwar, 2023-09-21:
The values of `scanFreq` and `headRotatePerSec`are being updated because you are supplying values for `verticalResolution` and `horizontalResolution` in your survey. If you remove those, the warning should disappear.

Also, the definition of a `verticalResolution` with a multi-beam scanner does not really make sense, as this is given by the spacing of the individual beams, and the `horizontalResolution` (if provided) would adjust for the `headRotatePerSec`.

Note that the use of these types of scanners (with multiple, fanned lasers, rotating around 360 deg) uses the _headRotation_ to achieve the 360 deg horizontal FoV and the _scanFreq_, which represents the vertical rotation in the case of a single laser beam, should be zero.

See the Velodyne VLP-16 ("Puck") example here:
https://github.com/3dgeo-heidelberg/helios/blob/a3909937432de147a276666b14db5794f5853845/data/scanners_tls.xml#L111-L130

Hannes9, 2023-09-21:
Okey, so as I understood correctly it makes sense in my case for removing the `verticalResolution` and the `horizontalResolution`.

And set the `scanFreqMin_Hz` and the `scanFreqMax_Hz `to "0" in the scanners.xml to have no vertical rotation and the `scanAngleMax_deg` and the `scanAngleEffectiveMax_deg` in the scanners.xml to "1" to have a full (360) horizontal roation, right ?

lwiniwar, 2023-09-21:
Yes, the `verticalResolution` for these scanners is governed by the multiple beams and their orientation.

The `horizontalResolution` is governed by the `headRotatePerSec_deg`, but if you give a specific value for the `horizontalResolution` it will start to overwrite some other settings.

Yes, just copy the settings from the Velodyne Sensor. As far as I understand the sensor you want to simulate, it performs very similarly.

Cheers,
Lukas

Hannes9, 2023-09-21:
Ok perfect. I'll change it.
How is then the `scanFreq_hz` value with 20 Hz in the survey.xml behaving as the range is set to 0 in the scanners.xml ß

Yes, probably it's similar. It's just they I am simulating two lidar sensor at its different positions (see [picture).

[image]

lwiniwar, 2023-09-21:
If `scanFreq_hz` is outside of the bounds of `scanFreqMin_hz` - `scanFreqMax_hz`, it should lead to an error, unless the upper bound is set to `0`, in which case all positive values are allowed:

https://github.com/3dgeo-heidelberg/helios/blob/a3909937432de147a276666b14db5794f5853845/src/assetloading/XmlSurveyLoader.cpp#L616-L647

As your `maxScanAngle_rad` is set to `1`, you might observe that the "lines" generated by the fanned out beams move vertically. I would set `scanFreq_hz` to zero to avoid that.

Ad the two scanners: yes, but that's more a question of the mounting than of the scanner itself ;)

lwiniwar, 2023-09-22:
Your two scanners have separate axes of rotation. You need do define this in every channel using

```<headRotateAxis x="0" y="0" z="1"/>```

... or you could also remove one of the scanners from your scanner.xml, run the simulation once, change the orientation in your platform, run the simulation again and then merge the point clouds.

Hannes9, 2023-09-22:
Exactly, but is this option in the channel settings supported ? I set the headrotateaxis to z and y for the respective channels, but I still get only some vertical points.

lwiniwar, 2023-09-22:
It might not be, @albertoesmp should know more. Maybe he can answer if he gets a chance.
In the meantime, I would suggest running two surveys after each other. Sorry for the complications!

albertoesmp, 2023-09-22:
In HELIOS++, multi-channel scanners are implemented as a collection of scanning devices. Each scanning device has its own scanner head. The `<headRotateAxis ...>` element defines the rotation of the scanner head. Therefore, when specified, a multi-channel scanner should support one arbitrary head rotation per device.

You can see how the `headRotateAxis` element is loaded here: https://github.com/3dgeo-heidelberg/helios/blob/a3909937432de147a276666b14db5794f5853845/src/assetloading/XmlAssetsLoader.cpp#L1563-L1582

Perhaps we can have a look at the XML snippet where the `headRotateAxis` is defined to understand how it is configured.

Hannes9, 2023-09-22:
Hi @albertoesmp,

Sure, I understand per one multi-channel device. In my case I would like to simulate 2 Multi-channel devices with the channels 1 to 31 (one device) and 32 to 62 (second device) -> see above previous Scanners.xml.

Here is a file snippet how I had it set now for both sensors with its different orientations:
`    <channel id="30">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="-19.5" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="31">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="-20.8" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
			
			
            <channel id="32">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="-19.5" />
				</beamOrigin>
				<headRotateAxis x="0" y="1" z="0"/>
			</channel>
            <channel id="33">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="-18.2" />
				</beamOrigin>
				<headRotateAxis x="0" y="1" z="0"/>
			</channel>`

albertoesmp, 2023-09-22:
I can confirm the `headRotateAxis` element works. I did some debugging for the following snippet, and the values are correctly updated (see the screenshot below showing the rotation axis for the three different heads):

```xml
<channels>
  <channel>
	  <headRotateAxis x="1" y="0" z="0"/>
  </channel>
  <channel>
	  <headRotateAxis x="0" y="1" z="0"/>
  </channel>
  <channel>
	  <headRotateAxis x="0" y="0" z="1"/>
  </channel>
</channels>
```		 

[image]

However, some parameters (like `headRotatePerSecMax_deg` or `headRotatePerSec_deg`) could interfere with the head rotate axis. These parameters can be set in the scanner file but overloaded by the survey file through the `scannerSettings` element. Just in case, make sure that these are not set to zero in your scanner and your survey. To be on the safe side, make sure whenever you are using `scannerSettings`, you also explicitly force these attributes to be non-null. In the snippet below, "..." can be replaced by your desired configuration, but don't forget to include the headRotatePerSec-related attributes, just in case.

```xml
<scannerSettings ... headRotatePerSecMax_deg="20" headRotatePerSec_deg="5"/>
```

If, despite these recommendations, the simulation is not working as expected, we will need either a minimum reproducible example or the entire use case to reproduce the issue and find what is going on.

Best,
Alberto

Hannes9, 2023-09-22:
OK thanks. I'll try it.
`headRotatePerSexMax_deg` and `headRotatePerSec_deg` I set both to 7200 in the s`cannerSettings` in the survey.xml file.
With headRotatePerSec-related attributes you mean the `headRotateStart_deg` and `the headRotateStop_deg`, right ?

albertoesmp, 2023-09-22:
I was thinking about the `headRotatePerSecMax_deg` and `headRotatePerSec_deg` but yes, the [start, end] interval also needs to make sense.

Hannes9, 2023-09-24:
Ah okey. Yes I set the `headRotatePerSecMax_deg="7200"` and `headRotatePerSec_deg="7200"` and the `headRotateStart_deg="0" `and the `headRotateStop_deg="360"` in the survey.xml. With those values the simulation is running to end until "0.00" time left, but somehow also no ouput with "simulation finished" is following which is unusual.

Somehow I also recognized the the `headRotatePerSecMax_deg` is not taken (see picture below). Mabye I have to set it in the scanners.xml instead of in the `scannersettings` in the survey.xml.

[image]

The `beamDivergence_rad` is not taken as well with its values. I'ts sometimes takes 0.003 instead of the provided 0.007.

han16nah, 2023-12-29:
Hi @Hannes9 
there haven't been any updates for a while, is this issue still open?

The `headRotatePerSecMax_deg` and the `beamDivergernce_rad` are parameters of the `scanner` of the **scanner XML**. If you are not sure, where a parameter has to be defined, you can check the summary tables at the end of the respective wiki pages.
You will find 
- `headRotatePerSecMax_deg` and `beamDivergernce_rad` on the [Scanner XML page](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#xml-tag-and-parameter-summary) 
- and `headRotatePerSec_deg`, `headRotateStart_deg` and `headRotateStop_deg` on the [Survey XML page](https://github.com/3dgeo-heidelberg/helios/wiki/Survey#xml-tag-and-parameter-summary).

And if not provided (or provided in the wrong location), the parameters take on some default value, e.g., 0.003 for the `beamDivergence_rad`.

I cannot explain why your survey did not finish. For this, we would need a minimum reproducible example.
Looking at the survey file you sent in the beginning (https://github.com/3dgeo-heidelberg/helios/issues/374#issuecomment-1729311050), please make sure to use dot (not comma) as decimal separator also for the coordinates of the legs. 

Best,

Hannah

Hannes9, 2023-12-30:
Hi @han16nah ,

I used all of these parameters now in the **Survey XML** except the `headRotatePerSecMax_deg` which I set in the **Scanner XML** and it is working. Therefor, this issue can is solved and can get closed now.

closed 2023-12-30 by han16nah