3dgeo-heidelberg/helios #375: Linux distribution not working

bhjolly, 2023-09-21:
**The problem**
Following instructions for downloading/running Helios++ on Linux does not work

**How to reproduce**
```
wget https://github.com/3dgeo-heidelberg/helios/releases/download/v1.2.0/helios-plusplus-lin.tar.gz
tar -xzvf helios-plusplus-lin.tar.gz
cd helios-plusplus-lin/
mamba env create -f conda-environment.yml
conda activate pyhelios_env
source heliosShell.sh
run/helios --help
```

**Expected behavior**
Help text for the helios executable

**Actual behavior**
```
$ run/helios --help
run/helios: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

Tried manually adding conda libs to LD_LIBRARY_PATH
```
$ LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$CONDA_PREFIX/lib run/helios --help
run/helios: error while loading shared libraries: libhdf5_serial.so.103: cannot open shared object file: No such file or directory
```

libhdf5_serial.so.100 exists in run/ so symlink that to .103 (hackey, yes I know) and get
```
$ ln -s libhdf5_serial.so.100 run/libhdf5_serial.so.103
$ LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$CONDA_PREFIX/lib run/helios --help
run/helios: error while loading shared libraries: libhdf5_serial_hl.so.100: cannot open shared object file: No such file or directory
```

At which point I gave up

**Your environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - HELIOS++/pyhelios version (or Git revision) that exhibits the issue: 1.2.0
 - Last HELIOS++/pyhelios version that did not exhibit the issue (if applicable): Not sure
 - Further system details (if relevant): i7 6700, 64 GB RAM

han16nah, 2023-09-21:
Hi @bhjolly 

you're on a good way with the LD_LIBRARY_PATH. Could you please try adding the `helios/run` folder instead of your `lib` folder?
Please also refer to our [release notes](https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.2.0) (at the very end), where we mention the necessary command.

Let us know if it still does not work and we will have a deeper look! 🙂 

Cheers,

Hannah

bhjolly, 2023-09-21:
Thanks Hannah, yep that's already handled when I source heliosShell.sh:
```
(pyhelios_env) jollyb@shali:/mnt/ext4_fast/singularity_builds/helios/testing/helios-plusplus-lin$ source heliosShell.sh
(pyhelios_env) jollyb@shali:/mnt/ext4_fast/singularity_builds/helios/testing/helios-plusplus-lin$ echo $LD_LIBRARY_PATH
:/mnt/ext4_fast/singularity_builds/helios/testing/helios-plusplus-lin/run
``` 
I notice the files in run/ have different version(??) numbers than what was expected, and the final missing file (libhdf5_serial_hl.so.100) isn't in there at all.

Happy to admit I could still be missing something though...

bhjolly, 2023-09-22:
Thanks Lukas

It turned out that setting LD_LIBRARY_PATH in my bash session wasn't enough, I had to export it as well. That solved all but one of my imports, I'm still stuck on libhdf5_serial_hl.so (nothing seems to provide it). Linking that to libhdf5_serial.so (just to see what happens) lead to a couple of errors with my symlinked libraries reporting the wrong version when helios tried to use them.

bhjolly, 2023-09-22:
I've actually gone down the route of building it all from source, following the steps in the Dockerfile. This seems to have worked well, but it's a bit frustrating that the release distribution didn't work so easily

lwiniwar, 2023-09-22:
We're glad you got it to work with the Dockerfile. Sorry about the release not working as we should, and thanks for your feedback. We'll do our best to fix it with the next release :)
Cheers,
Lukas

closed 2023-09-22 by lwiniwar

bhjolly, 2023-09-24:
@lwiniwar Thanks, let me know if I can contribute. I've also got a Singularity (/Apptainer) recipe similar to the Dockerfile that I can submit via PR if you think it would be useful (a few HPC facilities use this in lieu of Docker)