3dgeo-heidelberg/helios #376: Record pulses with no echos?

tim-devereux, 2023-10-03:
Hi there,  

In the wiki under [ASCII FWF format](https://github.com/3dgeo-heidelberg/helios/wiki/Output#ascii-fwf-format) I have read that -  "Only pulses that returned an echo are recorded". 

Would it be possible to have an option to record pulses that have no echos? This would be useful for the for canopy gap modelling simulations. 

Thanks.

albertoesmp, 2023-10-04:
Hi Tim,

Thank you for being interested in working with the full waveform data. We've been discussing this today in our meeting. If we understood correctly, you want to have the ASCII FWF output for any ray that intersects something, whether the returned energy passes the simulation filters or not. Is that correct?

If so, we would happily extend the simulator to account for this output. At the moment, we have a secondary compilation mode we use for debugging and analysis. When building HELIOS++ in this mode, we export extra data. We could extend this mode to export the additional data and send it back to you. If you are happy with the results, we can help you compile HELIOS++ to export this data.

Thank you again for your interest.

Best,
Alberto

tim-devereux, 2023-10-04:
Hi Alberto, 

Thanks for the reply. To be more clear, I'm after the rays that do not intersect anything. Specifically the staring position and the direction vector of those rays. 

- Tim

albertoesmp, 2023-10-05:
Ok, I think I got it. You want an output file with columns (ox, oy, oz, vx, vy, vz, c) where o are the coordinates of the ray origin, v is the director vector and c is a boolean flag specifying whether the subray was captured or not. Will this satisfy your requirements?

tim-devereux, 2023-10-05:
Actually, having the ray index and time info would also be useful. Maybe recording the uncaptured rays in the FWF ASCII, but with a null value for FWFIntensities would work? It could be activated with a command line argument.

Thanks, 
Tim

albertoesmp, 2023-10-09:
Hi again,

We have merged some new code to export this output (https://github.com/3dgeo-heidelberg/helios/commit/19c9007c9d570f36f9790a495acdb6f7f290fcb8). You must specify `--writePulse` when calling HELIOS++. If so, you will have files like "leg000_pulse.txt" in your output directory. The rows of this file should be (ox, oy, oz, vx, vy, vz, time, pulseIndex, deviceIndex). The (ox, oy, oz) should be the coordinates of the ray's origin, the (vx, vy, vz) the ray's director vector, the time when the ray was emitted expressed in nanoseconds, the index of the ray (pulseIndex), and the index of the scanning device (will always be the same unless you are using a multi-channel scanner simulated as many devices).

Note that the updates are available from now on in the dev branch (https://github.com/3dgeo-heidelberg/helios/tree/dev). Please let us know if it works as expected or if further modifications are needed.

Best,
Alberto

tim-devereux, 2023-10-10:
Thank you for adding this functionality Alberto, it will be key for our research on TLS plant area retrievals. 

One other request - Is it possible to update the Docker containers available on Dockerhub or test the Dockerfile? I cannot seem to successfully build an image.

myermo, 2023-10-11:
Hi @tim-devereux:

How are you trying to build / pull the image?

I just successfully built the repository image without problems using the provided Dockerfile, and I can pull any image from the Dockerhub without problems. Also, I checked that Helios++ is working inside the containers built from those images.

Best,
Miguel

tim-devereux, 2023-10-12:
Hi @myermo, 

Thanks for looking into this. I am using image_builder.sh with python3.8 and the dev branch. I get this error. [docker_build_error.txt](https://github.com/3dgeo-heidelberg/helios/files/12876069/docker_build_error.txt)

The images on Dockerhub seem to be out of date, the last commit in the dev container is from 2022. It would be good to have these images on Dockerhub set to [auto build](https://docs.docker.com/docker-hub/builds/) whenever changes are pushed to GitHub. That way we can have the latest changes without having to compile on our own machines. I attempted to do this with my own branch but could not get past the error shown above.

myermo, 2023-10-12:
Hi,

indeed, a funny error. When I wrote the script, the default version of Python installed by Conda was 3.8, exactly the one used by default in the script. For that reason, the compilation at that time was flawless. Now, Conda install Python 3.11, so we must provide to the Helios++ configuration the path where the python installed by Conda is located.

Should be solved now. Wait a few minutes before dev branch is updated.

Thank you for taking the time to send us the log.

Best,
Miguel

tim-devereux, 2023-10-18:
Thanks Miguel, It's working as expected now.

lwiniwar, 2023-11-03:
@tim-devereux can this issue be closed or does the output not work as expected? 
Thanks!

closed 2023-12-29 by han16nah

tim-devereux, 2024-07-28:
Hi @albertoesmp 

Apologies for reviving this thread, but I was wondering if you could add functionality to set --writePulse in pyhelios? 

Thanks, 
Tim