3dgeo-heidelberg/helios #389: Preprocessing step in Helios somehow possible to avoid big amount of data?

Hannes9, 2023-10-27:
E.g. removing some data already during mapping (reflections, fringe points, etc.). For example, scanners provide two measurements for each point and then we keep only one of them.

Thanks a lot!
Hannes

lwiniwar, 2023-10-27:
Hi @Hannes9 ,
I'm not sure what you mean. You want to filter the resulting simulated point cloud before export?

- Reflections: HELIOS++ does not handle reflections, so no reflected points are ever recorded
- Scanners provide two measurements - do you mean, multiple returns? You can set `maxNOR` to 1 in the scanner settings (see the wiki: https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#xml-tag-and-parameter-summary)

Cheers,
Lukas

Hannes9, 2023-11-22:
Hi Lukas.

I set the `maxNOR` to 1, but it is somehow not taking the value. Instead it takes still 0. Maybe it is not matching with my 	`pulseFreqs_Hz`	= "640000" ?

Hannes9, 2023-11-23:
Sure. 
Scanner settings:

	<scanner 	id 							= "hesai_XT32M2X"
				accuracy_m					= "0.01"		
				name              			= "Hesai XT32M2X 32-Channel Medium-Range Mechanical Lidar"
				optics            			= "rotating"
				pulseLength_ns     			= "30"
				rangeMin_m					= "0.05"
				rangeMax_m					= "300"
				pulseFreqs_Hz				= "640000"
				scanAngleMax_deg   			= "360"
				scanAngleEffectiveMax_deg	= "360"
				scanFreqMin_Hz				= "5"
				scanFreqMax_Hz				= "20"
				wavelength_nm				= "905"
				maxNOR						= "1"
				headRotatePerSecMax_deg 	= "7200"> 				
				
		<FWFSettings beamSampleQuality="1"/>	

		<channels>
		
			<channel id="0">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="19.5" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="1">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="18.2" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="2">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="16.9" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="3">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="15.6" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="4">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="14.3" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="5">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="13.0" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="6">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="11.7" />
				</beamOrigin>
				<headRotateAxis x="0" y="0" z="1"/>
			</channel>
            <channel id="7">
				<beamOrigin x="0" y="0.0305" z="1.9">
[... 734 lines cut ...]
			<scannerSettings template="scanner1"/>
		</leg>
		<leg>
			<platformSettings template="platform1" x="2697515.250000" y="5349204.500000" z="472.029999" />
			<scannerSettings template="scanner1"/>
		</leg>
		<leg>
			<platformSettings template="platform1" x="2697517.500000" y="5349202.500000" z="472.029999" />
			<scannerSettings template="scanner1"/>
		</leg>
		<leg>
			<platformSettings template="platform1" x="2697517.250000" y="5349197.000000" z="472.910004" />
			<scannerSettings template="scanner1"/>
		</leg>
		<leg>
			<platformSettings template="platform1" x="2697518.750000" y="5349197.000000" z="472.910004" />
			<scannerSettings template="scanner1"/>
		</leg>
		<leg>
			<platformSettings template="platform1" x="2697519.250000" y="5349199.500000" z="472.204987" />
			<scannerSettings template="scanner1"/>
		</leg>
	</survey>
</document>
```

Hannes9, 2023-11-23:
Addittionally, I would like to ask you if you can confirm that the Multi-Scanner & Multi-Channel scanner (see in the picture) is correctly set up in the Scanner settings XML ? Mainly regarding the positions (rotations/translations) of the two LiDAR sensors (rotating sensors), so that the virtual scanning in helios is very similar/ the same. 
[image]

Thanks a lot !

Hannes9, 2023-12-05:
Hi @lwiniwar,

Thanks already for your answer, that helps a lot !
I would have a few questions regarding your comment:

- Why is the `headRotatePerSec_deg="3600"` and not 360 ? 
- Does the total rotation time and the time of completing the trajectory needs to be same or can I set a really high value for  `headRotateStop_deg="36000"/>` -> see in example: 
```
 <platformSettings x="-40" y="0" z="0" movePerSec_m='8' /> <!-- with 8 m/s, we make it to +40 in 10 seconds -->
			<scannerSettings template="scaset" trajectoryTimeInterval_s="0.01" headRotatePerSec_deg="3600" headRotateStart_deg="0" headRotateStop_deg="36000"/> <!-- the rotation will also take 10 sec
```
- Thanks for the picture. When looking a the cube, what are the circled lines on left side of the circle ? Is it due that the LiDAR sensor on top is slightly placed in the lateral plane of the cube ?`

Thanks !

lwiniwar, 2023-12-05:
Hi @Hannes9 ,
1) with `headRotatePerSec_deg="3600"` you get 10 Hz "scan rate", i.e., 10 full scans of 360 deg field of view every second. If you set it to 360, you get one full 360 deg scan per second.
2) Setting `headRotateStop_deg` to a very high value will result in a very long scan and lots of points as recorded from the end of the leg. So I would suggest to calculate the actual runtime needed for the leg.
3) In this survey, the scanner starts at (-40, 0, 0) and the dimensions of the cube are (-50, -50, -50) to (+50, +50, +50). So you are seeing the "bottom" lines of the vertically mounted scanner resulting in these circles. Here's another screenshot after about 3/5th of the first full turn:
[image]

Cheers,
Lukas

Hannes9, 2023-12-05:
Hi @lwiniwar ,

1. But as we are moving in our cube 10 seconds in total, there should be 100 full rotations with 10 Hz, right ? So your first picture is without moving in the cube or ?
2. If I have multiple legs, I need to approx. calculate the runtime for every single leg ?
3. Thanks for the info !

lwiniwar, 2023-12-05:
Hi @Hannes9,

1. Yes, with the configuration as I posted it, there are 100 rotations. I just displayed the point cloud after the first rotation (after the first 0.1 seconds).
2. Yes, the values are per leg. If you have start and end position and the speed (which are both set in the survey XML), this should not be an _approximation_ but the exact value (up to float resolution). I agree that this is a bit impractical but currently the only workaround to allow for fan-like scanners.

Cheers,
Lukas

Hannes9, 2023-12-06:
Hi @lwiniwar,
1. Thanks !
2. Yes, I managed to get & set the values for all my legs.

In general, the virtual scanned data looks much better now. The problems that I now have is **on one hand** that the **simulation is not at all fully finishing** (maybe due to RAM limitations) and **on the other hand** that it looks like that the **file size would be extremely big.**

I think reducing the scene & legs won't be a good solution for me to consider as I need the scene and the legs how I defined them. Is there any way, e.g. to set the `movePerSec_m` for instance to 100 and recalculated certain values (e.g. `headRotatePerSec_deg` to 360000 you get 1.000 Hz "scan rate", i.e., 1000 full scans of 360 deg field of view every second,  `headRotateStop_deg` for each leg) to keep the same realistic scanning conditions as with  `movePerSec_m=1` ? Or would it result at the end to a complete different scan, similar file sizes & simulation time ?

Thanks !

lwiniwar, 2023-12-06:
Hi @Hannes9,

glad to hear we are making progress!

1. I think we should talk a bit about your expectations of the output before we see what parameters we can adapt/change. How long is the total runtime for your all of your legs now? Multiply that by the 640000 Hz Pulse repetition frequency you have in the scanner (- and, I believe, also by 32x2 channels. In fact, I am not sure if this value is set correctly.) and you will get the expected point count. If you don't use the `--lasOutput` flag yet, I suggest you do, since las files will be much smaller than txt/xyz-Files. 

Side note: if you increase the movement speed, and adapt all other parameters accordingly (head rotation, pulse frequency), this will result in the same point cloud (just with different `GPSTime` values). If you don't adapt e.g. the pulse frequency, you will get a smaller point cloud but then the results are not according to the specification anymore.

3. If the scene loads and starts scanning, it is unlikely to be a RAM issue, but it might be an issue with writing the output. Try if `--lasOutput` also solves this issue, and if not, if you can create a minimum example (one leg, as simple as possible scene) where it happens. We will probably have to run it on one of our machines to fully see what's going on here. You might also want to monitor RAM in the Task Manager while it is running up to the crash. Sorry we can't pinpoint it more exactly at this time.

Cheers,
Lukas

lwiniwar, 2023-12-06:
Hi @Hannes9,

I just double checked the results from yesterday and the 10 second survey results in 12 Mio. points. This corresponds to a total pulse repetition frequency of 1.2 MHz, which is not unheard of but probably not realistic for the scanner you are using (I can't seem to find the datasheet right away).
For each 'channel', this would be a PRR of 18.75 kHz. 

Cheers,
Lukas

Hannes9, 2023-12-06:
Hi @lwiniwar,

I need a .xyz file as output, that's for sure. Actually, how can I check for the expected total runtime ? And that you mean the multiplying like in the following: total runtime * 640000 * 64 (32*2) = expected point cloud size GB ?

The channels I set up with the help of the following pages of the LiDAR sensor manual: https://web2019.blob.core.windows.net/uploads/XT32M2X_User_Manual_X03-en-230110.pdf

[image]
[image]
[image]

Hannes9, 2023-12-06:
"...10 second survey results in 12 Mio. points. This corresponds to a total pulse repetition frequency of 1.2 MHz,..."
Maybe you should consider that I am using the sensor twice ? -> Data Points Generated Single Return: 640,000 points/sec (out of Sensor Manual) -> I have 2: 1,280,000 points/ sec -> *10sec -> 12,800,000 points in total

lwiniwar, 2023-12-06:
Hi again @Hannes9 ,

the datasheet says that it _generates_ 640,000 pts/second for a single one of the sensors. HELIOS++ expects you to provide a value valid for a single channel, which would be 20,000 Hz. Then it also doesn't matter if you have one sensor with 32 channels or two with - in total - 64 channels, because each of these channels operates with 20 kHz.
I see in the survey XML and the scanner XML that you never actually set `pulseFreq_hz`, only `pulseFreqs_hz`, which is supposed to be a list of possible options (and a warning given if the `pulseFreq_hz` is not one of the valid choices of `pulseFreqs_Hz`). 
So if you set `pulseFreq_hz="20000"` in the scanner setting tag of the survey XML, or as the only choice in `pulseFreq_hz`, it will be applied to all channels. This should reduce the number of points you are creating (and hence, the filesize) by a factor of 32. Additionally, the runtime should decrease by a similar factor, although we do have some overhead there.

regarding the output.xyz format: No, we don't support leaving out single columns there. The reasoning is that the xyz output is mainly for debugging and small scenes/point clouds, and we suggest the detour via las for any production use. With `las2txt` you can select what columns you want to export from the data.

So, first, I would fix the `scanFreq` and then, if you still run into issues, go via las.

Cheers,
Lukas

Hannes9, 2023-12-06:
Hi @lwiniwar,

Thanks, I will try it accordingly and let you know about my results and if it will be fine then with a .xyz output or I have to switch to las. 
I am sorry that I wasn't aware of this: _"HELIOS++ expects you to provide a value valid for a single channel, which would be 20,000 Hz."_ 

Maybe this question got lost, but how do I actually check for the expected total runtime ?

Thanks !

lwiniwar, 2023-12-06:
Hi @Hannes9 ,

No, it's not possible to predict the runtime for a simulation apart from very rough heuristics.

Re: the second option:
When you start the simulation, there is an internal counter that keeps track of all dynamics, e.g. the rotation of the scanner head, the laser pulses sent out, and the movement of the platform. This counter is written to the point cloud as "gpstime".

I hope this is clearer?

Cheers,
Lukas

lwiniwar, 2023-12-06:
Hi @Hannes9 ,
sure:

say you start a survey at 10:02:00 am computer time. It runs until 10:04:20, so a total time of 2 mins and 20 seconds. This is what I refer to as _runtime_, and what you would like to predict before running (not possible, see above).

In the survey, there is a single leg that the platform takes 10 seconds to run. This is what I call _simulation time_. If you don't set it specifically, the first point will have a GPSTime value of _the computer time when you started it (10:02:00 am) converted to GPS Time_ and the last point will have a GPSTime that is 10 seconds larger. So e.g., if the GPSTime of the first point is 414219, the GPSTime of the last point will be 414229. This time is also written to the trajectory file and can be used to connect the trajectory and the point cloud.

Cheers,
Lukas

Hannes9, 2023-12-06:
Hi @lwiniwar.

Ok, I understand the differences now. Thanks.
But what do you mean with connecting the trajectory and the point cloud ? How is it possible to connect it ?

lwiniwar, 2023-12-06:
Hi @Hannes9,
if you want to know the position of the platform (the trajectory) for when a specific point was recorded, you take the GPS Time of that point, find the temporally closest two entries in the .trj file and interpolate linearly between them. This allows you to do things like attribute each point with a beam vector, recalculate polar coordinates, ...

Cheers,
Lukas

Hannes9, 2023-12-06:
Hi @lwiniwar,

Ok, got it.
Thanks a lot !

Hannes9, 2023-12-06:
The change of the pulse frequency looks very good. Thanks a lot !

closed 2024-04-11 by lwiniwar