3dgeo-heidelberg/helios #395: Synchronization Issue with Visualization of MCAP Files Converted from XYZ

jeremy-fgn, 2023-11-01:
**Description:**
I am currently transitioning our simulation/debugging pipeline from Rviz to Foxglove studio and have started working with their open-source file format, MCAP. I've successfully modified the provided code snippet to convert xyz files into MCAP files for visualization purposes.

**Problem:**
I want to enhance the visualization by adding new features, but I'm encountering synchronization issues. Specifically, I want to:
1. Visualize "what the car sees over time" at a specified frequency (fps, one frame being one pointcloud).
2. Display "where the car was when it saw this."

The xyz files I have only contain `gpstime`, representing where the car was when it sent the beam. Computing when the beam should have returned using the speed of light and distance is an option, but hardware variability may make this computation inaccurate.

**Possible Solution:**
One approach could be having a separate timestamp along with `gpstime`, starting at 0 for the first entry of both files. Given the total simulation time, this timestamp could be divided by the number of entries in each file. 

**Example:**
Consider this toy example:
- xyz file:
```
x y z gpstime
1 1 0 0.056
40 1 0 0.052
5 1 0 0.060

```
- trajectory file:
```
x y z gpstime
50 50 50 0.052
55 50 50 0.056
60 50 50 0.060
```

We know the car's location at `gpstime` 0.052 and that it sent a beam at that time. However, without another clock, it's challenging to determine what the car received at this `gpstime`.

**Question:**
Is there a simpler, effective method to synchronize these files accurately, or does the proposed solution seem adequate? Your insights would be highly valuable.

lwiniwar, 2023-11-03:
Hi @jeremy-fgn ,

if I understand you right, you are concerned with the temporal mismatch between when a laser pulse is sent vs. when it is received.

Let's assume a maximum range $r$ of 1.5 km. Then, the time delay between these two measurements is 
$$
\Delta T = 2 \cdot r / c
$$
with $c = 300,000 $km/s.

Hence, $\Delta T$ turns out to be 1/100,000 s, or `1e-6`s (one nanosecond). The typical recording interval for `gpstime` (as also per your example) is `1e-3`s (one millisecond), so three orders of magnitude higher than that. I don't believe the signal round-trip time makes any difference for your visualization purposes.

If that's not what you referred to, please specify your problem differently.

Cheers,
Lukas

jeremy-fgn, 2023-11-03:
I agree with you. Still, is there a proper way to get **total simulation time** ? The way I'm doing it right now is by finding the min and max values for gpsTime.

I need this to have control over the fps when playing back the file.

albertoesmp, 2023-11-03:
Hi Jeremy,

Thank you again for your interest in HELIOS++.

If I understand you correctly, you want to know the elapsed time between the first instant in the simulation and the last one. In that case, you could look at the final lines of the standard output. You should see something like the following:

```
Elapsed simulation steps = 4000006
Elapsed virtual time = 13.3334 sec.
Main thread simulation loop finished in 2.77403 sec.
```

The simulation steps are the number of iterations computed by the simulator, the virtual time is the elapsed time in the simulation domain (I think this is what you want), and the main thread time is the execution time. 

Best,
Alberto

jeremy-fgn, 2023-11-05:
Thanks Alberto, that's indeed what I want. I'll try to make it work with pyhelios.

jeremy-fgn, 2023-11-10:
@albertoesmp I'm using pyhelios, and I basically want to be able to visualize each pointcloud sent by the lidar as a series of pointcloud sent over time. So if scanFreq_hz=50 , I want 50 pointclouds per second. I haven't found a straightforward way to do that in pyhelios, so currently I do the following:

1- manually parse the xml survey file to get `scanFreq_hz`

```
import xml.etree.ElementTree as ET

def getScannerFrequencies(survey_path):
    # Parse the XML file
    tree = ET.parse(survey_path)
    root = tree.getroot()

    frequencies = []

    for leg in root.findall(".//leg"):
        scanner_settings = leg.find("scannerSettings")
        if scanner_settings is not None:
            scan_freq = scanner_settings.get("scanFreq_hz")
            if scan_freq is not None:
                frequencies.append(int(scan_freq))
            else:
                raise ValueError(
                    f"No scanFreq_Hz found in scannerSettings for leg: {leg.attrib}"
                )
        else:
            raise ValueError(f"No scannerSettings found for leg: {leg.attrib}")

    if not frequencies:
        raise ValueError(
            "No scannerSettings with scanFreq_Hz found in the survey file."
        )

    return frequencies
```

2- try to get access to the **Elapsed virtual time** displayed in the console, from pyhelios, but it doesn't work:
```
    if simB.isFinished():
        print("\nSimulation has finished.")

    ElapsedTime = simB.stepLoop.getCurrentTime()
```

> 'SimulationBuild' object has no attribute 'stepLoop'

(taken from [src/sim/comps/SimulationReporter.cpp](https://github.com/3dgeo-heidelberg/helios/blob/c9f5eb7a79bbaa1fc8d7fef64b01fd4cbea59154/src/sim/comps/SimulationReporter.cpp#L39) )

I guess for now I'll resort to finding the min/max of gpsTime to know the duration of the simulation.

closed 2023-11-10 by jeremy-fgn