3dgeo-heidelberg/helios #396: Lib issues with 1.3.0 on Ubuntu

celine-mercier, 2023-11-02:
Hello,

We are trying to upgrade to the new helios version but ran into some lib linking issues.
We can probably find ways around it, but we were wondering what glibc version you have been compiling with? We suspect the issues come from us maybe running an older version, maybe because we are still on Ubuntu 20.04.

Best
Celine

han16nah, 2023-11-03:
Hi @celine-mercier,

thanks for reporting the issue! Can you give a few more details and send the error message(s)?

We left out some libraries in the distribution which we thought were not needed, specifically:
- libc.so.6
- libgcc_s.so.1
- libm.so.6
- libpthread.so.0
- libresolv.so.2
- libstdc++.so.6

However, if you're having issues because one of these is missing or there are conflicts, we will consider including them!

The `helios` executable is actually compiled with Ubuntu 20.04. We use the GitHub artifacts, so compilation is exactly like in the [workflow file](https://github.com/3dgeo-heidelberg/helios/blob/dev/.github/workflows/cmake.yml).

I have tested v1.3.0 on Ubunutu 22.04 where we have this libc.so.6:
```
~/Downloads/helios-plusplus-lin$ /lib/x86_64-linux-gnu/libc.so.6

GNU C Library (Ubuntu GLIBC 2.35-0ubuntu3.4) stable release version 2.35.
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 11.4.0.
libc ABIs: UNIQUE IFUNC ABSOLUTE
For bug reporting instructions, please see:
<https://bugs.launchpad.net/ubuntu/+source/glibc/+bugs>.
```

```
~/Downloads/helios-plusplus-lin$ file /lib/x86_64-linux-gnu/libc.so.6

/lib/x86_64-linux-gnu/libc.so.6: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=a43bfc8428df6623cd498c9c0caeb91aec9be4f9, for GNU/Linux 3.2.0, stripped
```

Again, thanks for the important feedback.

tik0, 2023-11-17:
Hi @celine-mercier and @han16nah 

I've made my first experiences with helios and I had also a linking problem on my fresh vanilla Ubuntu 20.04 and helios 1.3.0.
Maybe those insights can help you with that (btw. `python` links to `python3.8`):

First of all, I've checked that the `LD_LIBRARY_PATH` is empty and I also checked the linkage of `python`.
Everything looks fine:
```bash
~/Downloads/helios-plusplus-lin/run$ printenv LD_LIBRARY_PATH
~/Downloads/helios-plusplus-lin/run$ ldd $(which python)
        linux-vdso.so.1 (0x00007ffe277b0000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7980417000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f79803f4000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f79803ee000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f79803e9000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f798029a000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f798026c000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f798024e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f7980651000)
```
Then I sourced the heliosShell file, such that the `run` folder is appended to the  `LD_LIBRARY_PATH`.
This causes the issue, that the linking does no longer work for the python executable:

```bash
~/Downloads/helios-plusplus-lin/run$ ..
~/Downloads/helios-plusplus-lin$ source ../heliosShell.sh
~/Downloads/helios-plusplus-lin$ cd -
~/Downloads/helios-plusplus-lin/run$ python
~/Downloads/helios-plusplus-lin/run$ ldd $(which python)
/bin/bash: symbol lookup error: /bin/bash: undefined symbol: dlopen, version GLIBC_2.2.5
```

Adding the native OS library path fixed the linking issue and I can now import pyhelios as expected:

```bash
~/Downloads/helios-plusplus-lin/run$ export LD_LIBRARY_PATH="/lib/x86_64-linux-gnu:${LD_LIBRARY_PATH}"
~/Downloads/helios-plusplus-lin/run$ ldd $(which python)
        linux-vdso.so.1 (0x00007ffe277b0000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7980417000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f79803f4000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f79803ee000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f79803e9000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f798029a000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f798026c000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f798024e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f7980651000)
~/Downloads/helios-plusplus-lin$ python
Python 3.8.10 (default, May 26 2023, 14:05:08)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyhelios
>>>
```

dokempf, 2025-01-13:
With `v2.0.0` released and the completely reworked build system, I will go ahead and close this.

closed 2025-01-13 by dokempf