3dgeo-heidelberg/helios #397: Question about how trajectories are treated

jeremy-fgn, 2023-11-14:
Hi,
In a recent email exchange, I was told the following:

> At the moment, we are using a linear interpolation based on Euler's method y(t+h) = y(t) + h dydt(t). From this, we split the trajectory into regions that can be approximated like that. If you want to model a far-from-linear function, the model will use more linear partitions. If the function has linear-like regions, it will use fewer partitions. Until now, this has been enough for all use cases because, typically, trajectory files have enough points per second. Extending the interpolated trajectories to work with polynomial splits, splines, parametric interpolation, or other methods is easy. We also have Expression Trees, so we could even link them to the platform to govern its behavior with equations, e.g., (x(t) = 5*cos(t), y(t) = 5*sin(t), z(t) = sin(10*t)/20) to have a circumference-like trajectory on the plane with small periodic perturbations on z. Alternatively, we could extend interpolated platforms with an Expression Tree to model the perturbations. For example, the platform position can be modeled as an interpolated trajectory. Thus, if interpolated platforms are insufficient, let us know how to extend them to fit your use case. This is not problematic from the software's perspective.

The way I currently create non-linear trajectories involves a script that creates a piecewise cubic spline trajectory given a few reference points. From this fitted spline, the script then generates a dense set of around 500 x,y coordinates along the smooth trajectory, suitable for plotting and exporting in a .trj file.

[image]

For now, if I give this .trj file to HELIOS++, it will simulate the LiDAR moving in many small straight lines connecting those 500 x y coordinates - is my understanding correct ?

If I want to make the simulated trajectory smoother, should I:
- generate a denser set of x y coordinates along the smooth trajectory? Will this impact simulation time by a lot?
- request for smooth trajectory support?

Thanks.

albertoesmp, 2023-11-14:
Hi Jeremy!

Yes, you are correct. We initially thought about two or three strategies for the interpolation but then we noted that for real trajectories we often have a lot of points so a linear model is more than enough. I think that increasing the resolution will give you a more smooth trajectory. I don't think it will have a significant negative impact on performance unless you use many points. You are using 500 points, you could easily use 5000 and expect no significant increase in computation time. In case you have problems with the memory, you can play with the `slopFilterThreshold` parameter as described [here](https://github.com/3dgeo-heidelberg/helios/wiki/Survey#interpolated-trajectories).

If the above doesn't work for you, we can also extend the simulator with another interpolation strategy. Let us know if this is necessary.

jeremy-fgn, 2023-11-14:
I guess linear interpolation should be enough then, thanks!