3dgeo-heidelberg/helios #398: Cannot run survey with custom platform and interpolated trajectory

jeremy-fgn, 2023-11-14:
**The problem**
Cannot run survey with custom platform and interpolated trajectory.

**How to reproduce**
1- Remove the line that launches pyhelios in the xmlchecker script
2- launch xmlchecker on the survey file - **raises an error**
4- Run pyhelios SimulationBuilder on said file - **raises another error**

**Expected behavior**

Error 1:

- I expected the survey file and its dependencies to pass the xmlchecker verification

Error 2:

- I expected the trajectory file to be good for interpolation.
- I could not manage to specify a model/mesh for the scanner, even when I tried to make the id and name of the platform match the name of the .obj file.

**Actual behavior**
xmlchecker error:
```
urllib.error.URLError: <urlopen error [WinError 2] The system cannot find the file specified: 'C:\\[redacted]\\helios-plusplus-win\\interpolated'>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "helios-plusplus-win/pyhelios/util/xmlchecker.py", line 72, in <module>
    assert str(platform_file) == "interpolated"
AssertionError
```

pyhelios error:
```
Using default value for attribute 'trajectory_separator' : ,
ERROR: Failed to read platform asset definition: configs\survey_cones_increasing_distance.xml#interpolated
EXCEPTION: invalid unordered_map<K, T> key
Execution aborted!
```

**Environment:**
 - OS: Windows 10
 - HELIOS++/pyhelios version: HELIOS++ VERSION 1.3.0 (helios-plusplus-win)

**Files**
[tree_structure.txt](https://github.com/3dgeo-heidelberg/helios/files/13355787/tree_structure.txt)
[configs.zip](https://github.com/3dgeo-heidelberg/helios/files/13355836/configs.zip)

albertoesmp, 2023-11-14:
Hi there!

I think there is a problem with the trajectory files. I can see that the elements in the header are separated with (",") but those in the body are separated with (" "). This format issue might be the reason behind the trajectory separator issue. I transformed the *trajectory_linear_increase_3s.trj* file to use (","). See it attached in this answer. Could you check 
if this solves the problem?

[trajectory_linear_increase_3s.trj.zip](https://github.com/3dgeo-heidelberg/helios/files/13356746/trajectory_linear_increase_3s.trj.zip)

jeremy-fgn, 2023-11-15:
It solves error 2 (pyhelios error above), thank you for that.

Error 1 is still raised, independently of running xmlchecker or not:
```
urllib.error.URLError: <urlopen error [WinError 2] The system cannot find the file specified: 'C:\\[redacted]\\helios-plusplus-win\\interpolated'>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "helios-plusplus-win/pyhelios/util/xmlchecker.py", line 72, in <module>
    assert str(platform_file) == "interpolated"
AssertionError
```

I'm guessing I'm doing something wrong in the survey file. I was trying to replicate [trajectory interpolation](https://github.com/3dgeo-heidelberg/helios/wiki/Survey#interpolated-trajectories) for my use case.
I want to set a basePlatform to set my LiDAR at a certain height. But I also want the trajectory to be interpolated.

han16nah, 2023-11-15:
Hi @jeremy-fgn,
there is nothing wrong with the survey file.
For me, it works when I remove the comment from the `platform.xml`, right in line 2. I am not sure, why this is an issue but can you confirm that this solves your problem?
Best,
Hannah

jeremy-fgn, 2023-11-15:
It works! But xmlchecker still raises an error - here's the [xmlchecker_errorlog_for_interpolated_platform.txt](https://github.com/3dgeo-heidelberg/helios/files/13367371/xmlchecker_errorlog_for_interpolated_platform.txt) if you wanna look into it.

So I guess two small bugs need to be fixed:
- commenting the `platform.xml` file caused an issue for pyhelios - ideally it shouldn't
- xmlchecker raises an error if `platform="interpolated"` - it seems like it's looking for a file or platform id literally named `interpolated`

Let me know if I should close this issue or not. Thanks for the support as always.

fafraobait, 2024-06-24:
Hey @han16nah, 
I think I might be getting a related error.

> HELIOS++ VERSION 1.3.0
> 
> CWD: "/workspaces/helios-docker"
> seed: AUTO
> surveyPath: ".survey.xml"
> assetsPath: "assets/"
> outputPath: "output/"
> writeWaveform: 0
> writePulse: 0
> calcEchowidth: 0
> fullWaveNoise: 0
> splitByChannel: 0
> parallelization: 1
> njobs: 0
> chunkSize: 32
> warehouseFactor: 4
> platformNoiseDisabled: 0
> legNoiseDisabled: 0
> rebuildScene: 0
> lasOutput: 0
> las10: 0
> fixedIncidenceAngle: 0
> gpsStartTime: 
> kdtType: 4
> kdtJobs: 0
> kdtGeomJobs: 0
> sahLossNodes: 32
> 
> xmlDocFilename: .survey.xml
> xmlDocFilePath: 
> xmlDocFilename: my_scanners.xml
> xmlDocFilePath: 
> Using default value for attribute 'averagePower_w' : 4
> Using default value for attribute 'beamQualityFactor' : 1
> Using default value for attribute 'opticalEfficiency' : 0.99
> Using default value for attribute 'receiverDiameter_m' : 0.15
> Using default value for attribute 'atmosphericVisibility_km' : 23
> Using default value for attribute 'wavelength_nm' : 1064
> Using default value for attribute 'headRotatePerSecMax_deg' : 0
> Using default value for attribute 'rangeMax_m' : 1.79769e+308
> Number of subsampling rays (0): 19
> Using default value for attribute 'scanFreqMin_Hz' : 10
> Using default value for attribute 'scanFreqMax_Hz' : 250
> Using default value for attribute 'scanAngleEffectiveMax_deg' : 180
> Using default value for attribute 'accuracy_m' : 0.005
> Using default value for attribute 'rangeMin_m' : 1
> Using default value for attribute 'rangeMax_m' : 1.79769e+308
> Using default value for attribute 'headRotatePerSecMax_deg' : 0
> Using default value for attribute 'beamDivergence_rad' : 0.0005
> Using default value for attribute 'pulseLength_ns' : 4
> Using default value for attribute 'maxNOR' : 0
> Number of subsampling rays (1): 19
> Using default value for attribute 'scanFreqMin_Hz' : 10
> Using default value for attribute 'scanFreqMax_Hz' : 250
> Using default value for attribute 'scanAngleEffectiveMax_deg' : 180
> Using default value for attribute 'accuracy_m' : 0.005
> Using default value for attribute 'rangeMin_m' : 1
> Using default value for attribute 'rangeMax_m' : 1.79769e+308
> Using default value for attribute 'headRotatePerSecMax_deg' : 0
> Using default value for attribute 'beamDivergence_rad' : 0.0005
> Using default value for attribute 'pulseLength_ns' : 4
> Using default value for attribute 'maxNOR' : 0
> Using default value for attribute 'trajectory_separator' : ,
> ERROR: Failed to read platform asset definition: /.survey.xml#interpolated
> EXCEPTION: _Map_base::at
> Execution aborted!
> terminate called after throwing an instance of 'HeliosException'
>   what():  ERROR: Failed to read platform asset definition: /.survey.xml#interpolated
> EXCEPTION: _Map_base::at
> Execution aborted!
> Aborted (core dumped)

As @jeremy-fgn described, it seems that it's looking for `interpolated` in the survey file.

I'm using the Linux release. Do you know how to circumvent this?

Files to reproduce the error:
[files.zip](https://github.com/user-attachments/files/15951149/files.zip)

albertoesmp, 2024-06-24:
Dear @fafraobait ,

I've tried your survey and found a problem when using the header-like specification. I've already corrected it, but it will not be available until the next release (also, you will need to remove the space between # and H, i.e., #HEADER instead of # HEADER in the trajectory file).

For now, you can run your simulation by specifying the data represented by each column directly in the XML, like below:

```xml
<platformSettings 
  trajectory=".traj.trj" toRadians="false"
  tIndex="0" xIndex="4" yIndex="5" zIndex="6" rollIndex="1" pitchIndex="2" yawIndex="3"
/>
```

Thank you for reporting this!

fafraobait, 2024-06-24:
Thanks @albertoesmp, seems to be working now!

closed 2024-11-28 by han16nah