3dgeo-heidelberg/helios #401: Canceled simulation

Hannes9, 2023-11-25:
Hi :)

When I reduced the `movePerSec_m` from 20 to e.g. 5 or 1, it is somehow canceling my running simulation without any error message (see image).
What could be the problem causing this ?

I was thinking of the solution to split my simulation into less legs, but the better way would be to be able to run the simulation completely successfully.

[image]

Thanks a lot !

Hannes9, 2023-11-27:
Hi.

Sure, here are my files. I hope that you can open it properly. Unfortunately, the scene parts I can't provide.

[mls_-1_data.zip](https://github.com/3dgeo-heidelberg/helios/files/13471904/mls_-1_data.zip)

[scanners_als.zip](https://github.com/3dgeo-heidelberg/helios/files/13471897/scanners_als.zip)

[platforms.zip](https://github.com/3dgeo-heidelberg/helios/files/13471895/platforms.zip)

Scene XML: 
```
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <scene id="tum_scene_-1" name="TUM_-1_Scene">
<!--         <part>
            <filter type="objloader">
                <param type="string" key="filepath" value="data/sceneparts/basic/groundplane/groundplane.obj" />
            </filter>
            <filter type="scale">
                <param type="double" key="scale" value="80" />
            </filter>
			<filter type="translate">
				<param type="vec3" key="offset" value="2697480;5349175;470" />
			</filter>
			<filter type="rotate">
				<param key="rotation" type="rotation">
					<rot angle_deg="-10" axis="z"/>
				</param>
			</filter>
        </part>  -->
        <part>
            <filter type="objloader">
                <param type="string" key="filepath" value="anonymized_obj_file.obj" />
            </filter>
			<filter type="rotate">
				<param key="rotation" type="rotation">
					<rot angle_deg="0" axis="z"/>
				</param>
			</filter>
			<filter type="scale">
                <param type="double" key="scale" value="1" />
            </filter>
			<filter type="translate">
				<param type="integer" key="onGround" value="-1" />
				<param type="vec3" key="offset" value="0;0;0" />
			</filter>
        </part>
    </scene>
</document>
```

Thanks a lot !

han16nah, 2023-11-27:
Hi @Hannes9 

I have tested your survey (with a simple groundplane, translated towards your trajectory point and scaled x1000). For me, it runs through (in 57 minutes).
Maybe this is a memory-related problem. How large is your .scene file?

Hannes9, 2023-11-27:
Hi.

It runs through with `movePerSec_m` = 1 ?
My free storage on my PC is 103 GB where Helios is installed and 746 GB free where the output is saved.
My .obj scene file has a size of ca. 4,5 GB

han16nah, 2023-11-27:
Yes, exactly, but my .scene file has a size of 1 KB 😅 
Did it run through with `movePerSec_m = 20`?
How much RAM does your machine have?

Hannes9, 2023-11-27:
Ah okey, good.
Yes it the simulation run completely with `movePerSec_m = 20`.
I have 16 GB RAM.

Hannes9, 2023-12-06:
Hi @han16nah,

Thanks for the info.

I am now chatting with @lwiniwar in the task "Preprocessing step in Helios somehow possible to avoid big amount of data? #389" about this topic as well. It's mainly about how to reduce the resulting file sizes with manipulating the scanning conditions, but still keep the same realistic conditions as for `movePerSec_m = 1`. E.g. my total point cloud file should have a size maybe between 5-15 GB, but for now with `movePerSec_m = 1` with not even finishing the first leg, the file has already around 11 GB 😅.

I think we can just continue in the task that I mentioned in the beginning and we can close this task.

closed 2023-12-26 by han16nah