3dgeo-heidelberg/helios #407: Missing leg in the generated point cloud using simulation

AJAY31797, 2023-12-28:
Hi, 

I am trying to generate point cloud from a 3D model. The surveys file I am using is as follows:
`<?xml version="1.0"?>
<document>
	<scannerSettings id="template1" active="true" pulseFreq_hz="100000" scanAngle_deg="45"/> 
	<survey name="building_scan_1" platform="data/platforms.xml#copter_linearpath" scanner="data/scanners_als.xml#livox_mid-70" scene="data/scenes/Buildings/building1.xml#building_scene">
	
		<leg>
			<platformSettings  x="-65.0" y="-48.0" z="115.000" onGround="false" movePerSec_m="5"/>
			<scannerSettings template="template1" trajectoryTimeInterval_s="0.05"/>
		</leg>
		<leg>
			<platformSettings  x="65.0" y="-48.0" z="115.000" onGround="false" movePerSec_m="5"/>
			<scannerSettings template="template1" trajectoryTimeInterval_s="0.05"/>
		</leg>
		<leg>
			<platformSettings  x="-65.0" y="38.0" z="115.000" onGround="false" movePerSec_m="5"/>
			<scannerSettings template="template1" trajectoryTimeInterval_s="0.05"/>
		</leg>
		<leg>
			<platformSettings  x="65.0" y="38.0" z="115.000" onGround="false" movePerSec_m="5"/>
			<scannerSettings template="template1" trajectoryTimeInterval_s="0.05"/>
		</leg>
	</survey>
</document>`

After running the simulation, I am getting four outputs with the name of four legs as shown in the image. 
[image]

On opening the legs, the fourth file, i.e. leg003.xyz shows nothing. It says there is nothing to load. The size of this file is also 0KB. I am not sure why this is happening. Is there something wrong with the survey file, or is it because all the points are already covered in the first three legs, so there is no new point in leg003? 

Best Regards,
Ajay

albertoesmp, 2023-12-28:
Hi Ajay,

When using a trajectory made of linear legs, for example four legs L0, L1, L2, and L3, you will have points for the trajectory L1-L0, also for L2-L1, and for L3-L2. However, for L3 you will not have points because there is no trajectory vector (i.e., L4-L3 is not possible because there is no L4). It might be different if you use a scanner with a static position that rotates. In that case you will have points for each leg because each legs completes a given angular interval.

I hope this solves your doubts. Please, let us know if you have further questions.

Best,
Alberto

AJAY31797, 2023-12-28:
Hi Alberto, 

Many thanks for your quick response. 

Best Regards,
Ajay

closed 2023-12-28 by AJAY31797