3dgeo-heidelberg/helios #410: Generation of GroundTruth from total .obj file

Hannes9, 2024-01-09:
Hi,

Is there any possibility to generate a Ground Truth point cloud for the imported .obj file ? So, for example a setting that a scanner is able to reach certain occluded points/ point sampling with the scanner settings.

Thanks a lot.

lwiniwar, 2024-01-09:
Hi @Hannes9 ,
If I understand you correctly, you want a simulation where occlusion is ignored and all objects are - in addition to generating returns - translucent and allow the ray to continue after intersection?
This is not possible in HELIOS++, sorry.
If you just want points sampled randomly on all surfaces of your .obj, you can use external software like cloudcompare.
Cheers, Lukas

Hannes9, 2024-01-09:
Hi @lwiniwar ,
Exactlly. True, thanks a lot for the info !

closed 2024-01-09 by Hannes9