3dgeo-heidelberg/helios #412: Reason for noisy point cloud output in Helios ?

Hannes9, 2024-01-24:
Hi,
What could be the reason that my partial output point clouds in .xyz format from Helios are a bit noisy in comparison to the actual Ground Truth point cloud ?
And you have a recommandation how to avoid that in advance or post process ?

Thanks !

[image]
[image]
[image]
[image]

han16nah, 2024-01-24:
Dear @Hannes9 

as documented on the page [Error sources and randomness control](https://github.com/3dgeo-heidelberg/helios/wiki/Error-sources-and-randomness-control), HELIOS++ supports different sources of error to mimic real laser scanning systems.

The main one is the **ranging error** of the sensor.  In your scanner XML, there is a parameter `accuracy_m` which corresponds to the standard deviation of the ranging measurement. If not set, a default value of 0.003 m is used. You can try setting it to 0.0 and see if you get the desired output.
(In addition, noise can be added to the platform or to the full waveform (see Wiki), but these are probably not set in your experiment)

Furthermore, you could color the point cloud by return number and investigate if the noise comes from multiple returns, caused by the virtual beams hitting the surface at steep incidence angles or by partial hits, i.e., only part of the laser beam footprint is hitting a surface. You could avoid multiple returns by setting the `maxNOR` (maximum number of returns) in the scanner XML or you could play around with the `beamSampleQuality` which controls how the beam footprint is approximated using subrays (see [Fullwave](https://github.com/3dgeo-heidelberg/helios/wiki/Fullwave)).

Lastly, your problem could also be related to coordinate precision.

We can give you further hints if you provide more details:
- info about the scenario you are simulating in your experiments (and what output you expect)
- configuration XML files
- output point clouds.

Also, do the last two images show only the simulated point cloud or both your "ground truth" and the simulated point cloud on top of each other? In case of the latter, please visualize the point clouds with different colors and also add a scale bar, which can help us to interpret what the "problem" is.

Cheers,

Hannah

Hannes9, 2024-01-24:
Dear @han16nah,

Thanks for providing this information. This is already quite useful.
To your bullet points:

- the scenario is mainly an indoor scenario with many tiny pipe segments and objects etc.
- which configuration XMLs do you need specifically ? All ?
- The resulted Point Clouds I can not share with you unfortunately due to a NDA

The last two images shows both.
Red point cloud is Ground Truth. 
[image]

han16nah, 2024-01-25:
If you still don't get desired results after adapting the `accuracy_m` (for the scanner, i.e., in the scanner XML) and maybe the `beamSampleQuality` (in the survey XML in the `FWFSettings`-tag), then please send us your XML files. Especially **scanner** and **survey** are important, the scene would also be useful (can be a dummy scene with the toyblocks in the HELIOS++ distribution).

However, looking at the scale of the point cloud you sent, this level of noise is quite realistic for laser scanning point clouds. So if you try to mimic real laser scanning acquisitions very closely, excluding noise completely is not the way to go. If you are only interested in some aspects of the laser scanning acquisition, e.g., occlusion, then it is of course valid to disable noise/ranging errors.

Hannes9, 2024-01-25:
Hi @han16nah,

It worked with changing the` accuracy_m` to 0.0. 
Thanks a lot for the help !

closed 2024-01-25 by han16nah