3dgeo-heidelberg/helios #413: Interactive UI

Hannes9, 2024-02-08:
Hi,
how is it possible to set up an interactive UI for the scanning simulations shown in one of your videos: [https://www.youtube.com/watch?v=1SOg7b5q4ak](https://www.youtube.com/watch?v=1SOg7b5q4ak) ?

Thanks a lot!

han16nah, 2024-02-09:
Dear @Hannes9 

we removed this visual UI from from HELIOS when we ported the code to C++ in favour of runtime and maintainability. This decision was made to focus more on features and improvements relevant for researchers.
The simulations can be visualized with the script `run/helios.py`. For advanced users, HELIOS++ can be built with a visual debug mode: https://github.com/3dgeo-heidelberg/helios/blob/dev/BUILDME.md#install-on-linux-with-visual-debug

Cheers,

Hannah

Hannes9, 2024-02-09:
Dear @han16nah,

Thanks for the quick response.
I understand, thanks for the information.

Best regards,
Hannes

closed 2024-02-09 by han16nah