3dgeo-heidelberg/helios #416: Graphical launcher for HELIOS++

dg-505, 2024-02-20:
Hi,

just wanted to mention that I've created a small graphical launcher for HELIOS++ using Qt5:
https://github.com/dg-505/helios-launcher

It's still in an early alpha status, so for now you can choose the HELIOS++ installation directory, the survey path, and set options via a text field. From that the command line is executed and the output is displayed in a text window.

The plan is to expand the functionality in the future (but it's a free time project for me).

Feel free to take a look, and if you find it helpful, you probably might consider mentioning it somewhere so that others may possibly benefit from it.

And if you think this really sucks and wouldn't be of any use, just let me know, then I probably drop it (or probably not).

Cheers and have a nice day
Jonathan

han16nah, 2024-02-23:
Dear Jonathan @dg-505 

thank you for these cool news. We tested the helios launcher and it works as expected, great work!
We are always happy about contributions and am sure that other users will find the helios launcher useful as well. We link to the launcher [in our README](https://github.com/3dgeo-heidelberg/helios?tab=readme-ov-file#gift-related-projects-and-contributions), is it okay like that?
We are always happy to discuss usability improvements. Small teaser in this regard: We will soon be working on a high-level Python API that will make it much easier to configure, execute, and visualize HELIOS++ simulations, as part of our project [Sustainable HELIOS++](https://www.geog.uni-heidelberg.de/gis/sustainableHELIOS_en.html).

Cheers,

Hannah

closed 2024-02-23 by han16nah

dg-505, 2024-02-23:
Yeah thank you, I like that! 👍 
Thanks for the teaser.

Cheers
Jonathan

dg-505, 2024-03-18:
Hi there,
I've just released a new version of the graphical launcher.
We now can set the optional arguments for HELIOS++ by using interactive GUI elements.
Feel free to have a look.

Cheers
Jonathan