3dgeo-heidelberg/helios #423: [Suggestion] 2D beam divergence leading to elliptical footprint

dg-505, 2024-02-28:
**Is your feature request related to a problem? Please describe.**
Currently the beam divergence is set via the `beamDivergence_rad` attribute in the `<scanner>` tag. This implies a circular footprint cause the beam diverges in all directions by the same amount.
However, as you might know, some (if not many) airborne scanners have an elliptical footprint with a beam divergence of e.g. `1.6 x 0.5 mrad` [RIEGL miniVUX-3UAV data sheet](http://www.riegl.com/uploads/tx_pxpriegldownloads/RIEGL_miniVUX-3UAV_Datasheet_2022-09-27.pdf)

**Describe the solution you'd like**
Would it be possible to extend the laser beam simulation to simulate an elliptical footprint?

lwiniwar, 2024-02-29:
Hi @dg-505 , 
thanks for the suggestion. We have thought about this in the past.
Sampling rays uniformly along ellipses is, unfortunately, an elliptic integral problem with only approximate solutions. We have put this off for the time being as we continue to improve the simulation for circular footprints.
We plan to take the results from that and see if we can implement an approximation for the elliptical case. Until then, you can run two simulations - one with 0.5 and one with 1.6 mrad - and see if it actually makes a difference in your case.

Cheers,
Lukas

dg-505, 2024-02-29:
Thank you :+1: