3dgeo-heidelberg/helios #425: How o correctly add attributes to a LAS output

waitforraining, 2024-03-07:
I have already used stripID to write the data returned by multiple legs into one LAS file to avoid processing multiple times. At the same time, I also need to add an index similar to scanIdx in the LAS file. I modified the compiled code, but the result of running it is not correct.
How should I correctly modify this part of the code to add attributes to the LAS output?
in LasWriterSpec i add scannerAttrIdx and scannerAttrStart 
[image]
[image]
[image]
in LasMeasurementWriteStrategy
[image]
[image]
After this,i use laspy to read las.Even though I already have the 'scanidx' attribute, the attribute values are incorrect.

albertoesmp, 2024-03-07:
Hi!

How are you configuring the `scannerType`? If you look in the addExtraAttributes, you can use 9 for double or 5 for int, as done with the other extra attributes. I'd also recommend having a look at the LASTools examples (https://github.com/LAStools/LAStools/tree/master/LASlib/example) to find something close to what you want. Then, you can compare it with your implementation to check that you are not missing any steps.

In case you don't manage to extend the LAS output as you want, you might also extend the `filems::write::strategies::DirectMeasurementWriteStrategy` to export your output in text format. Then you can check that it is correct. If it is not, maybe the problem is the data itself and not the writing. Also, you can always convert the ASCII point clouds to LAS/LAZ using LASTools.

Thank you for using the HELIOS++ simulator, and let us know if you have further issues.

waitforraining, 2024-03-08:
Hi!
I am a rookie in C++ and made a mistake. I should use legIdx to tag point,but i use scanIdx in pointMeasurement. Learn from you response,I revise code,and get correct legIdx like this.
[image]
it is funny.
thanks you!

closed 2024-03-08 by waitforraining