3dgeo-heidelberg/helios #427: Scanner setting for Leica BLK 360 Laser Scanner

thivageran97, 2024-03-13:
Hi, i'm currently working on a project that uses the BLK 360 Laser scanner. I'd be interested to know if anyone has created a setting for the following scanner. I have tried creating my own setting according to the spec sheet but I'm unsure of what most of the setting does and what is the exact value based on this scanner. Apart from that, I would like to know if there is a setting to define the density of the points or like the resolution of the points for instance with the blk at 10m distance you have like x amount of points on an given area and u can change the point resolution to have more points on a particular surface at 10m distance. Other than that, i do realize the beam travels to a certain distance is there a way to set the maximum distance of this laser rays?

han16nah, 2024-11-28:
Hi @thivageran97 

have you implemented the Leica BLK 360 yet? If you still need our help, please send the XML scanner settings you created and are unsure about and we will have a look.

Cheers,

Hannah

thivageran97, 2024-11-28:
Hi, I have managed to adjust the settings but I'm not sure if it replicates 100 percent exactly like the actual scanner. I'm happy to share my scanner setting and probably you could advise on how i would be able to change it. 

	
	<scanner 	id 							= "blk360"
				name              			= "BLK 360"
				accuracy_m					= "0.008"
			 	beamDivergence_rad 			= "0.0004"
				headRotatePerSecMax_deg		= "30"
				
				optics            			= "rotating"
				pulseFreqs_Hz				= "1440000"
				pulseLength_ns     			= "4"
				rangeMin_m					= "0.6"
				scanAngleMax_deg   			= "180"
				scanAngleEffectiveMax_deg   = "150"
				scanFreqMin_Hz				= "2"
				scanFreqMax_Hz				= "120">
				wavelength_nm				= "830"
		
		<FWFSettings beamSampleQuality="3"/>	
		<beamOrigin x="0" y="0" z="0.2">
			<rot axis="y" angle_deg="0" />
			<rot axis="z" angle_deg="0" />
			<rot axis="x" angle_deg="0" />
		</beamOrigin>
		<headRotateAxis x="0" y="0" z="1"/>
		
	</scanner>

han16nah, 2024-11-28:
Hi,

according to the specifications, I would say you need to set the `scanAngleMax_deg` to 135° (since the full vertical field of view is 270°). And probably also the `scanAngleEffectiveMax_deg`? But I am not sure about this.
And the `pulseFreqs_Hz` to 680000 Hz (or some list with this value being the maximum). You may also set the `rangeMin_m` to 0.5 and `rangeMax_m` to 45. 

The user selectable resolutions are 6/12/25/50 mm @ 10 m = 0.034°, 0.069°, 0.143° and 0.286°. From this, you can theoretically calculate the scan speed and head rotation speeds that are needed or you directly provide the vertical and horizontal resolutions to HELIOS in the survey file. Let us know if you obtain the desired point spacing. 

Do you have this sensor available so that you can compare the simulated point cloud to the expected real point cloud on a simple test acquisition (e.g., scanning a wall)?

thivageran97, 2024-11-29:
Hi, 

Sure for the scananglemax I'm not sure either i will just try 135 and see if that works. 

Otherwise for my survey setting this is what i use at the moment:

<scannerSettings id="profile1" active="true" pulseFreq_hz="100000" scanFreq_hz="120" scanAngle_deg="100" headRotatePerSec_deg="10.0"/>
    <survey name="tub_1" scene="data/scenes/my_scene/ifc_tub1.xml#ifc_tub1" platform="data/platforms.xml#tripod" scanner="data/scanners_tls.xml#blk360">
        <FWFSettings binSize_ns="0.2" beamSampleQuality="3"/>
        

Should i change the pulse frequency on my survey to 6800000Hz as well? Apart from that, shall i also change the speed because I'm using 10 at the moment. but if based on the different settings how can i change them? 

Yes, I would be able to borrow one to test it out. by test you mean place the scanner let say x distance from a wall and try to have a model which has the same size as a room and see how different they are?