3dgeo-heidelberg/helios #452: Abnormal Simulated Intensity Values

NanBlanc, 2024-05-30:
Hi 3DGeo team,
I just tried the current dev version of HELIOS++ (GitHash: 4d50c542) 
and figured out that intensity values return are abnormal (ranging from 0.0001 to 0.0007) compared to expected ones (around 500, depending of the setup)

Here are the graphs of the values distributions from the same simulation : 
[image]
Using current dev version of HELIOS++ (v2) (GitHash: 4d50c542) 

[image]
Using current stable version of HELIOS++ v1.3.0 (GitHash: 79df55e8)

Also in the current dev version (v2), the argument --version still returns `HELIOS_1.3.0_REL_pybind_dynBoost`, you might want to change it to avoid confusions

albertoesmp, 2024-05-30:
Dear @NanBlanc, thank you very much for your observation. We are currently working on alternative energy models, which is why you get different results. If you want to use the dev branch with the old energy model, run with the `--legacyEnergyModel` flag to enable it.

NanBlanc, 2024-05-30:
Thanks @albertoesmp, i will stick to the stable one until you release a stable for the new energy models :)

han16nah, 2025-03-25:
Will close this issue, since this bug only occurred in a development version due to unfinished implementations of new energy models.

closed 2025-03-25 by han16nah