3dgeo-heidelberg/helios #468: No Hit Tiny Targets

suptimq, 2024-08-03:
Hi,

I have a bunch of branch obj files and a trunk obj file and would like to run the TLS simulation. However, only one or two branches and the trunk could be retrieved after the simulation. After checking https://github.com/3dgeo-heidelberg/helios/issues/441 and https://github.com/3dgeo-heidelberg/helios/issues/376#issuecomment-1746624827. I wondered what the best way is to make sure I can get those tiny branches hit.

I have attached the `survey` and `scene` xml files as well as a few obj files for reproducing this.

https://drive.google.com/drive/folders/1s0dDtr1Lz-HmoAVLMnMR5P_Mv9Qy_q8m?usp=sharing

han16nah, 2024-08-03:
Dear @suptimq 

as you already saw, we had similar reports recently with very small faces not being scanned. We will investigate the issue and come back to you. Thanks for reporting and thanks for the demo data!

Best wishes,

Hannah

han16nah, 2025-01-31:
Hi Tim @suptimq 

It seems like we finally fixed this with #517 🥳 
We just released a new minor version v2.0.2 which you can install with `conda install -c conda-forge`.
Do you want to check if your tree can now be scanned completely with HELIOS++?

Cheers,

Hannah

closed 2025-01-31 by han16nah