3dgeo-heidelberg/helios #507: Exploring the Integration of Zenmuse L2 LiDAR into Helios++

juanjosegonzalezresearch, 2025-01-17:
We are currently working with the Zenmuse L2 LiDAR for real-world scanning applications and are interested in simulating these scans within Helios++. Would it be possible to include this LiDAR model in the database of LiDAR parameters available in Helios++? We are fully prepared to collaborate with your team to facilitate its integration.

lwiniwar, 2025-01-22:
Hi @juanjosegonzalezresearch and thank you for your interest in integrating the DJI L2 scanner into HELIOS++. As you also mention in issue #508, the inner workings of this scanner are not fully documented but we've been trying to engineer a solution and believe we will have a prototype soon.
However, this prototype is implemented in `python` and we are currently lacking the expertise to port and include this code in helios' `C++` codebase.
If this is something you could potentially help with? If so, let's get in touch!

juanjosegonzalezresearch, 2025-02-14:
Hi, @lwiniwar.

Thank you so much for reaching out and for the great work you’re doing on integrating the DJI L2 scanner into HELIOS++. It’s really exciting to hear that you already have a prototype in development!

At the moment, I’m fully committed to another project, so unfortunately, I won’t be able to assist with the C++ implementation right now. However, I’d love to contribute in the future when my schedule allows. 

Wishing you all the best with the next steps, and looking forward to staying in touch.

Thank you so much. Best.

closed 2025-08-07 by dokempf