3dgeo-heidelberg/helios #508: Exploring unknown noise discrepancies between Real-World scan and Helios++ simulation

juanjosegonzalezresearch, 2025-01-21:
We have conducted a real-world scan of a kind of open-box and simulated the same open-box in Helios++ (Figure 1). Both the real-world scan and the simulation were performed using parameters as closely matched as possible: the same flight paths, material properties, mesh dimensions, etc. The real-world scan was conducted with the Zenmuse L2, while the simulation in Helios++ utilized a modified Livox Mid 40 sensor. This modification involved adjusting its beam divergence: the modified Livox Mid 40 retains all the parameters of the original Livox Mid 40 but adopts the beam divergence of the Zenmuse L2.

It is likely that the parameters of the modified Livox Mid 40 do not perfectly correspond to those of the Zenmuse L2 due to limited access to detailed specifications. Additionally, we are aware that Helios++ simulates circular divergences rather than elliptical ones. Nevertheless, we observed a significant level of noise in the real-world scan that is absent in the simulation (Figure 2).

Might there be factors not currently accounted for in Helios++ simulations that could help explain this discrepancy? If there are any aspects of our simulation setup that we may have misconfigured, we would greatly appreciate your guidance to help us accurately reproduce the observed noise.

[image]
[image]

lwiniwar, 2025-01-22:
Hi @juanjosegonzalezresearch ,

the interior walls appear in the point cloud because the subray-approach generates returns there (as there are mesh triangles that are being intersected with the rays). On the outside (what you labelled as "noise"), there are no mesh triangles. 
In the real point cloud, these appear as light is reflected by both the overhanging surface of the box and by the ground below, and they produce a combined return that results in somewhat of an "average" position of the two. Depending on how much of the footprint hits what surface, the average is closer to the box top or closer to the ground.

I'm not fully sure if we are able to reproduce that behaviour well with HELIOS++, but I can give you some pointers where I would start experimenting:
1) the pulse length of the laser. A longer pulse means that the two echoes will be overlapping more, and therefore subrays will be moving the maximum, which is used in the echo detectio
2) the bin size used for discretization. A larger bin size will result in a less sensitive maximum detection as two maxima cannot be seperated if they are assigned to the same bin
3) the window size used for maximum detection, to suppress secondary echoes.

It probably helps to think about the values and convert them to metric space using the speed of light, i.e., knowing that 1 ns corresponds to 30 cm distance; which is halved since the beam has to return to the laser scanner.

Looking at the echo number/return number might also offer additional insights, so if you recreate the figures you have shown above, coloring them by return number would be a good idea.

I hope this helps so far - if all of that changes nothing, you might have to look into changing the echo detection we currently have implemented.

juanjosegonzalezresearch, 2025-02-14:
Hi @lwiniwar.

Thank you very much for your detailed response and all the suggestions. I really appreciate the effort you've put into explaining many factors influencing this behavior.

I had already explored some of these aspects in my tests, but your insights provide me more angles to consider. Thanks.

Best.