3dgeo-heidelberg/helios #567: Simulation Capability for Riegl VUX-120 and Riegl VUX-160

GlenEaton, 2025-02-18:
Hi Helios ++ Team,
Thank you for providing such great software.
For my PHD and our project research we are using the Riegl VUX-120 23.
I would like to be able to simulate this system, but it seems currently not possible, since the rotating polygon has different shaped facets.  It would be great if this systems configuration could be implemented.
The Nadir configuration is standard, but the forwards and backwards scans are curved.
I would also be happy to assist if required.
I have attached the manufacturer datasheet.

Currently I have set up the configuration as 3 linear channels with 1/3 the pulse rate and scan frequency.  But I can only approximate a single tilt angle within the channels and need to adjust the angle to create a equidistant point spacing between the channels.
This is not how the Riegl system operates as the channels are not aligned and vary from 15 -10 degrees tilt across the swathe.

 Any assistance would be greatly appreciated.
Thanks for your help.
Regards,
Glen.

[RIEGL - 2024 - RIEGL VUX-120-23 datasheet.pdf](https://github.com/user-attachments/files/18840407/RIEGL.-.2024.-.RIEGL.VUX-120-23.datasheet.pdf)

han16nah, 2025-03-25:
Dear @GlenEaton 

Starting with your main question regarding the curved forwards and backwards scans towards the swath edges: We are afraid that our deflector models currently cannot replicate this. :( 

Besides that, your suggested approach of using 3 channels (-10°, 0°, 10°) with reduced pulse rates and scan frequency seems to be the way to go. But this will not replicate correct GPS Times for your returns, since in the simulation, pulses of the different scan lines would be sent out simultaneously, while for the real scanner, each scan line would be scanned consecutively. Alternatively, you could try running simulations with the same `--gpsStartTime` and the full PRR three times (-10°, 0, 10°) and combine the scans, keeping only every 3rd scan line from each using a custom script. Here, you would need to use the `pyhelios` point cloud output which contains the direction vector to determine the starts of a new scan line (in combination with the `GPSTime`?).  

(Ah, and looking at the datasheet, since the "Max. Effective Measurement Rate" is 2,000,000 (for PRR 2,400 kHz), you probably want to set your `scanAngleMax_deg` to 120 and `scanAngleEffectiveMax_deg` to 100)

@lwiniwar Feel free to correct me or add to my answer

GlenEaton, 2025-03-25:
Hi Han16nah,
Thanks for your feedback.
I have actually already implemented the new beam deflector required in a fork and it is working well.
In regards to the second method, I wasn't concerned about the GPS time so it was OK.  The main issue with that method is it doesn't have the varying forward backward scan angle across the swath.
In regards to the Max effective measurement rate, thanks for the comments.  I have decided what to do with that yet. I may just use the effective rate at the full scan and ignore the PRF, since I'm assuming the extra pulses account for the roll compensation.

My main problem now is getting the forward, backward, and nadir separate and the scan angles, since the Beam directions are global in the full wave form and I'd rather not have to use the output trajectory information to interpolate.

Regards,
Glen.