3dgeo-heidelberg/helios #587: Add writePulse flag to PyHelios

tim-devereux, 2025-03-13:
Description:
Currently, the PyHelios wrapper doesn't provide access to the --writePulse flag that exists in the command line version of Helios++.

- The --writePulse flag is available in the CLI version but cannot be set through the Python API
- This functionality would be valuable for workflows that require unbound pulse data analysis within Python

Thank you for considering this feature request.

Regards,
Tim

han16nah, 2025-04-10:
Hey Tim @tim-devereux 

I created a PR (#611) with a fix, it was quite straightforward.
Let me know if you are available to quickly test if it works as expected or if I should just go ahead and merge.
Thanks for using HELIOS++ and requesting features!

Cheers,

Hannah

tim-devereux, 2025-04-10:
Hi Hannah

Thanks for addressing this. I've tested it on our TLS simulations, everything looks good to me!

PR #611 (2025-04-11): enable writePulse in simulationBuilder (fixes #587)
This should enable using the `writePulse` flag from pyhelios (fixes #587)
commit e3d525f0 (2025-04-11): Merge pull request #611 from 3dgeo-heidelberg/feat/write_pulse_pyhelios

enable writePulse in simulationBuilder (fixes #587)
closed 2025-04-11 by han16nah