3dgeo-heidelberg/helios #600: `ori` in pyhelios measurement array is still shifted

han16nah, 2025-03-31:
With `pyhelios`, we currently get the following measurements output:

```
   0      1      2      3      4      5      6      7      8        9         10            11              12            13            14              15         16 
[pos.x, pos.y, pos.z, ori.x, ori.y, ori.z, dir.x, dir.y, dir.z, intensity, echoWidth, NumberOfReturns, ReturnNumber, FullwaveIndex, hitObjectId, classification, gpsTime]
```

where `ori` is the **origin** of the beam that created the return. However, with respect to the measured points and the input/output trajectory, these values are still shifted by the internal helios scene shift.

I validated this by plotting the origins and the trajectory for `data/surveys/custom_als_toyblocks.xml`, where the flight lines are configured from x="-50.0" to x="50.0" (or the other way around).

```python
output = sim.join()
measurements, trajectories = pyhelios.outputToNumpy(output)
ori = measurements[:, 3:6]
time = measurements[:, 16]
sc = plt.scatter(ori[:, 0], ori[:, 1], c=time)
plt.scatter(trajectories[:, 0], trajectories[:, 1], c='r', s=1)
plt.colorbar(sc)
plt.show()
```

[image]

And compared it to the helios shift that I got via:
``` python
scene = sim.sim.getScene()
shift = scene.getShift()
print(f"[{shift.x}, {shift.y}, {shift.z}]")
```

which returned `[20.0, 0.0, 10.889053]`

As expected at this point, `np.max(ori[:, 2])` is 69.810947 and not 80.0 as set as z-value in the survey file.

---

HELIOS++ / `pyhelios` version 2.0.2, Windows 11

GlenEaton, 2025-05-07:
Hi han16nah,
Does this issue still exist or has it been resolved?
Could you clarify which variables are affected please?
I see you have the origins and trajectories plotted.  The trajectories look OK, but I couldn't follow where the data came from (I dont use pyhelios).
Does it only happen in Pyhelios, or also in the output files from the commandline?
Is it related to the shift in the ground model, ie scene shift, or is there some other "internal helios scene shift" that is always applied.
I'm currently working with the commandline data, but may shortly move to running through pyhelios.
A few people our research group rely on the origins and beam directions in their computations, so hopefully they are correct.  
Thanks for your help,
Glen.

han16nah, 2025-05-08:
Dear Glen,

as far as I know, it only happens in pyhelios. In the command line, the beam origins are only exported when writing the full waveform, and there the shift is already applied correctly.

And yes, there is always an internal scene shift applied (using the centroid of the scene bounding box), so we do not run into problems when users provide a scene in geographic coordinate systems with very big coordinates.

I just created a PR for a possible solution to fix the pyhelios issue, see #622. Do you want to have a look and confirm? 

In general, are you and/or your colleagues willing to contribute to HELIOS++? We would really appreciate that and could discuss different options to get started :) 

PS: Unfortunately, I noticed another discrepancy in the GPS Time in the pyhelios trajectory output that we'd have to have a look at next.

GlenEaton, 2025-05-15:
Hi @han16nah,
Thanks for the response and sorry for not replying sooner.

I'm not that great with C++ yet and haven't really started with pyhelios, but I had a look at your fix and it seems reasonable if the shift isn't being reapplied in the c library before m.beamOrigin is called.  I did notice that in most of the C++ instances of the shift being applied it is simply called rather than reassigned, so my only question would be does the variable need to be reassigned or can you just directly call "shift" at this point.

In relation to contributing to HELIOS++ hopefully our group will do this as we progress.  There are already a few posts from people in our group in the issues page.  As a few of us are just starting to dive into Helios++ and pyhelios it may be a while before we can contribute on work outside our own projects, however we will definitely share any developments that we make ourselves that hopefully others are interested in.  There will definitely be more focus on Helios++ in the next 12-18 months from our group, so hopefully some contribution will come from that in relation to resolving issues and development.

Regards,
Glen.

GlenEaton, 2025-05-15:
Hi @han16nah,
I was just looking through the code and noticed in the write strategy listed below that the variable "shifted" is used for the measurement position output.  Maybe you could modify the cpp to call this variable instead.

C:\Users\user\helios\src\filems\write\strategies\DirectMeasurementWriteStrategy.h

`// ***  UTILS  *** //
    // *************** //
    /**
     * @brief Build a string from measurement data
     * @param m Measurement data itself
     * @param shift Shift for the measurement coordinates
     * @return String with measurement data
     */
    virtual std::string measurementToString(
        Measurement const &m,
        glm::dvec3 const & shift
    ){
        glm::dvec3 const shifted = m.position + shift;
        std::stringstream ss;
        ss << std::setprecision(4) << std::fixed;
        ss  << shifted.x << " "
            << shifted.y << " "
            << shifted.z << " "
            << m.intensity << " "
            << m.echo_width << " "
            << m.returnNumber << " "
            << m.pulseReturnNumber << " "
            << m.fullwaveIndex << " "
            << m.hitObjectId << " "
            << m.classification << " "
            << std::setprecision(9) << std::fixed
            << m.gpsTime / 1000000000.0 << std::endl;
        return ss.str();
    }`

Hope I making sense.
Regards,
Glen.

PR #622 (2025-05-15): Beamorigin fix
fixes #600 

When writing a test, I also noticed that the GPSTime in the pyhelios trajectory does not match the GPSTime in the pyhelios measurement (which corresponds to the GPSTime written to file, i.e., seems correct).
closed 2025-05-15 by han16nah