3dgeo-heidelberg/helios #631: Invalid GPS time of pyhelios trajectory output

han16nah, 2025-05-14:
The trajectory output of pyhelios seems to be invalid (see test for demonstration here: [https://github.com/3dgeo-heidelberg/helios/blob/f7eedfdad582ff1a17e4863c4349806adcd80af1/pytests/test_gpsStartTimeFlag.py#L70](pytests/test_gpsStartTimeFlag.py#L70))

See also screenshots below:

[image]

[image]

krzysztof-barden-hexagon, 2025-07-04:
im not sure if the rootcause is the same for an issue i had but here are my findings:
here the gps time is limited to first week of gpstime https://github.com/3dgeo-heidelberg/helios/blob/main/src/sim/core/Simulation.cpp#L266
in SurveyPlayback which inherits from Simulation the gps time is overwritten https://github.com/3dgeo-heidelberg/helios/blob/main/src/sim/core/SurveyPlayback.cpp#L90 so it can go beyond 1st week of gps time
then in doSimStep the check if we are still in 1st week of gps time is triggered https://github.com/3dgeo-heidelberg/helios/blob/main/src/sim/core/Simulation.cpp#L85 so basically every step subtracts a week from current gps time

han16nah, 2026-11-10:
Closing, since this bug does not persist in the current `alpha-dev` version

closed 2026-11-10 by han16nah