3dgeo-heidelberg/helios #658: XmlUtils::getAttribute with boost::get<double> but the default value is int causes exception

krzysztof-barden-hexagon, 2025-07-04:
an example:
https://github.com/3dgeo-heidelberg/helios/blob/02722bf490cb57ac9bd09a1141d780d262c57b92/src/assetloading/XmlAssetsLoader.cpp#L1019-L1020 but happens in multiple other places this happens in multiple places
this is because [ObjectT](https://github.com/3dgeo-heidelberg/helios/blob/main/src/assetloading/XmlUtils.cpp#L185) is a boost variant and when we pass an int, it gets initialized with [int](https://github.com/3dgeo-heidelberg/helios/blob/main/src/assetloading/XmlUtils.cpp#L223), so boost::get fails to find the double type and produces an exception

suggested solution: either fix all the places getAttribute is used or rewrite it so type errors cant happen

fexfl, 2025-07-29:
Hi, thank you for finding the error!

Since you mentioned that this happens in multiple other places, could you provide the other occurences that you found as well?
 In my brief search the example above was the only one I found with an incorrect default type. For all other calls of `XmlUtils::getAttribute` and `boost::get` the type matches, or the existence of a value in the xml file is required via `XmlUtils::hasAttribute`, such as here:
https://github.com/3dgeo-heidelberg/helios/blob/f490a05fa5c2b466d65f70fa2d4e4692f74f3001/src/assetloading/XmlAssetsLoader.cpp#L1142-L1150

Cheers,
Felix

krzysztof-barden-hexagon, 2025-07-29:
Hi, after double checking i do confirm this is now the only place. It might have that i checked an older version. 

I do suggest though to rewrite boost::get & xmlUtils::getAttribute combo into one method that would be type safe (produce compile error in case of type mismatch and remove need to provide type name in string form as an attribute. If you want we can cooperate in that matter.

closed 2025-08-06 by dokempf