3dgeo-heidelberg/helios #662: Sub_rays_info

UQ-Raja-Ram-Aryal, 2025-07-10:
**I would like to know the effect of partial hits on leaf area density estimation.**

Now we don;t have sub-rays level information

I want to get sub-ray levels information. 

Is there any way I can use pulse, fullwave and points data to get that information ?

If we have this, we can easily explain the impact of partial hit in any kind of leaf and wood surface and volume related works.

GlenEaton, 2025-07-17:
Hi Bernhard,
Thanks for the information on the sub-ray extraction.  I think Raja will look into that.

In regards to your response "it depends", I understand.  I realise that helios++ does things a little different to try and get a similar result to some of the more heavy simulation programs.  I'm doing analyses on UAV LiDAR simulations and came across the issues I am seeing when looking at the point density and distributions.  Unfortunately looking under the hood of Helios simulations is not the focus on my research, so hopefully you guys can do a systematic study.

My investigation is currently limited to confirming where the noise is coming from and mitigating it where I can.  At the end of the day I need to remember this is a simulation environment and I can't expect to use it to measure actual values from real data models.

Just my "two cents" which you can feel free to ignore, the weighting on the sub-rays is affecting the positioning.  All the sub-rays around the outside of the beam are overwhelming the center sub-ray.  Not just in the positioning but in the time difference or where the peak signal return is being interpolated.  So even on full hits when the sample quality is higher than 1, the position is off (within the level of beam divergence noise (actually by the amount of beam divergence at the distance from the scanner). Interestingly in the tests I ran, I get better results of position from partial hits than full hits (probably due to sub-rays being concentrated at one location of the footprint).

I have a small test.
You just need to change the beam sample quality between simulations.  Set to 1 and 3.
It uses the RIEGL VUX-1UAV22 system from the scanners_ALS.xml and linear copter from platforms.xml.
the cylinder scaled has a 60cm diameter.
If you view from top down on the points you can see the difference.  It helps to clip the top points off the cylinder also, so you only see the sides.

Toyblocks Data
Survey file
'''
<?xml version="1.0"?>
<document>
	<survey name="toyblocks_uls_VUX1_22" platform="data/platforms.xml#copter_linearpath" scanner="data/scanners_als.xml#riegl_vux-1uav22" scene="data/scenes/toyblocks/toyblocks_cylinder_scene.xml#toyblocks_cylinder_scene">
	
		<leg stripId="0">
			<platformSettings  x="-35.0" y="-5.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="true" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="0">
			<platformSettings  x="35.0" y="-5.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="false" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="1">
			<platformSettings  x="-35.0" y="5.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="true" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="1">
			<platformSettings  x="35.0" y="5.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="false" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="2">
			<platformSettings  x="-5.0" y="-35.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="true" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="2">
			<platformSettings  x="-5.0" y="35.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="false" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="3">
			<platformSettings  x="5.0" y="-35.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="true" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
		<leg stripId="3">
			<platformSettings  x="5.0" y="35.0" z="80.0" onGround="false" movePerSec_m="5"/>
			<scannerSettings active="false" pulseFreq_hz="1200000" scanAngle_deg="50" scanFreq_hz="93.5" trajectoryTimeInterval_s="0.01" />
		</leg>
	</survey>
</document>
'''
Scene
'''
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <scene id="toyblocks_cylinder_scene" name="toyblocks_cylinder_scene">
        <part>
            <filter type="objloader">
                <param type="string" key="filepath" value="data/sceneparts/toyblocks/cylinder.obj" />
            </filter>
			<filter type="scale">
                <param type="double" key="scale" value="0.03" />
            </filter>
        </part>
    </scene>
</document>
'''

My research is along the same lines as you have mentioned (pretty much exactly), specifically related to TLS and UAV-LiDAR 3D modeling and measure structure in trees. So when I get a little further along in my research I will definitely reach out for exchange of ideas or collaboration.  I do have some development I've added that will pull request in the future.

Anyway, thanks for your help.
Regards,
Glen.

GlenEaton, 2025-07-21:
Hi Han16nah,
Thanks for having a look.  Is the circular point plot with the binSize_ns set to 0.1, because this already seems like an improvement to what I was getting?
I think the outlying points are partial hits towards the outside of the cylinder parallel to the flight direction.  Need to use a single line to determine where the hit is on the cylinder.

I will run some tests again with the different binSize_ns and see how the data looks.
Previously I was just getting an overestimation of BeamSampleQuality 3. Your result looks more manageable for getting a realistic value and dealing with the noise.

I'll also have a look at the meshing scenario you mentioned.

Thanks for your help.
Regards,
Glen.