3dgeo-heidelberg/helios #672: Cannot create helios scene with 3 objects from Blender

crespopere, 2025-07-30:
Hi,

I'm new with HELIOS++, and also with Blender, and I'm trying to convert a virtual forest plot created with Blender to a TLS point cloud by using HELIOS++.

Firstly, I created the blender forest plot consisting of (i) ground, (ii) some trees located on the ground, and (iii) a big sphere envolving the forest plot. Each of these objects is stored in a separate collection named as "Ground", "Tree", and "Sphere", respectively. Additionally, there is a fourth collection named "Ignore", containing all I want to ignore. All these collections are within the "Scene Collection". Once I have these objects and structure I launch Blender2Helios without showing any error.

As a result, three folders are created within the "sceneparts" directory named as the previously mentioned collections. These folder contained an .obj and a .mtl file. Moreover, within the sceneparts I created a .mtl file name materials with the following content:

newmtl Ground
Ka 1 0 0
helios_classification 1

newmtl Tree
Ka 0 1 0
helios_classification 2

newmtl Sphere
Ka 0 0 1
helios_classification 3

On the other hand, .xml files are created in the scenes and surveys directories. I share the mentioned files through the following link: https://upvedues-my.sharepoint.com/:f:/g/personal/pabcrepe_upv_edu_es/EjGvbXGovbpFoYF9orrAoPsB2Utwr4MBoMA_Q3av56Fx4w?e=ShyHYr

The next step I tried was to generate the helios scene by launching the command "helios D:\Pablo\Estudios\2025_Maquette\Helios++\helios\data\surveys\blender2heliosScene.xml -vv" from a miniconda environment. However, only the objects ground and sphere are considered for the scene creation. Analyzing the command prompt output it shows for the Tree object: "

Reading 3D model from .obj file 'D:\Pablo\Estudios\2025_Maquette\Helios++\helios\data/sceneparts/Tree/Tree-2364-2786-1565.obj'...
Reading materials from .mtl file 'D:\Pablo\Estudios\2025_Maquette\Helios++\helios\data/sceneparts/Tree/Tree-2364-2786-1565.mtl'
1 material(s) loaded.
Unknown line ''
Error reading primitives.
EXCEPTION: bad allocation
.obj not found in cache
Adding primitive to Scenepart ...
# new primitives added: 0
# total primitives loaded: 0
Applying filter: rotate
Applying filter: translate
Applying filter: scale
XmlSceneLoader::validateScenePart detected an invalid scene part with id "2"
The path "D:\Pablo\Estudios\2025_Maquette\Helios++\helios\data/sceneparts/Tree/Tree-2364-2786-1565.obj" was given.
It leads to the loading of an invalid scene part, which is automatically ignored when composing the scene."

It has happened to me that I have been able to generate a scene with only trees, only sphere, only ground, trees + sphere, or ground + sphere, but never the 3 objects at the same time.

I would appreciate if you could help me out with this thing.

Thank you in advance,

Pablo.

han16nah, 2025-08-01:
Hi @crespopere 

I just tested your data and can reproduce the issue.

Can you please try the following: With a text editor, c**heck all your .OBJ files if there is a line `usemtl ` which is not followed by any value** (i.e., material name). It seems like HELIOS raises a `bad allocation` exception here because it tries to assign the second string in this line to the material name but there is none. Since the line occurs before all the face definitions (lines starting with `f`), the code breaks before reading any faces ("primitives"), resulting in the reported (0 primitives) and thus an invalid scene part.
If you **remove this line**, the sceneparts should be loaded.

Based on your description, it seems like you want to use the `materials.mtl` file for your sceneparts. In this case, you need to specify the relative path to this .mtl file in the beginning of your .OBJ file, e.g., replace `mtllib Ground-200-200-0.mtl` by `mtllib ../materials.mtl`. Furthermore, you need to specify the respective material name in the `usemtl` line, e.g., for the ground OBJ, replace the empty `usemtl` line by `usemtl Ground`. I hope this works!
Find more info on the OBJ (and MTL) format here: https://paulbourke.net/dataformats/obj/

Meanwhile, we will add a check to the HELIOS++ code to just ignore the `usemtl` line if no material name is provided and use the default material rather than breaking the OBJ file reading.

Thanks for reporting!

Cheers,
Hannah

crespopere, 2025-08-01:
Hi @han16nah ,

Thank you for your reply!

It worked by removing the usemtl line! Actually, I tried using materials.mtl since I thought the issue could be there.

Cheers,

Pablo